seandepagnier/autopilot_route_pi #10: Route Position Bearing: APB waypoint ID does not change

marcobergman, 2020-12-05:
Currently the destination waypoint id in the APB messages seem to be set to empty. This disables waypoint notifications at raymarine autopilots. I'd like to be able to switch the destination waypoint id's on, possibly by some config file setting. 

(btw I also tweaked my pypilot so it maintains heading at waypoint arrival until confirmed - based on changing waypoint id, consistent with raymarine)

Raspberry Pi 4 version 10 buster
OpenCPN 5.2.4
autopilot_route_pi-0.4.0.17-raspbian-armhf-10-buster-armhf/ from cloudsmith.io
also happens when compiled from source
testing method: https://github.com/marcobergman/ais_simulation

rgleason, 2023-01-22:
Do you have a screenshot that explains this.  Is it still happening? If not, it can be closed.

marcobergman, 2023-01-23:
It is still happening, and it can be seen as a minor thing: 

[image]

closed 2023-01-23 by marcobergman