seandepagnier/autopilot_route_pi #11: Advance Waypoint and Activate Waypoint missing

marcobergman, 2020-12-05:
Waypoint Bearing and standard XTE: Advance Waypoint and Activate Waypoint are not available anymore in the context menu. That is unhandy. Please maintain this functionality in this current plugin. And for bonus: a keyboard shortcut for Advance Waypoint would be brilliant.

rgleason, 2023-01-21:
These are available now, but please refer to the testing done here in Windows11  https://github.com/seandepagnier/autopilot_route_pi/issues/26

These functions appear to get frozen and don't work.

rgleason, 2023-01-22:
Marco, I think this is old and should be closed. You are the only one other than Sean who can close it.

rgleason, 2023-01-22:
Marco, can you confirm my recent posts about Issues?  If so, please add  a comment and say what OS and version.

marcobergman, 2023-01-23:
These options that are present in the stock Route Console are not present in the autopilot-route plugin:

[image]

closed 2023-01-23 by marcobergman