seandepagnier/autopilot_route_pi #7: Route Position Bearing: Console XTE, BRG and RNG no useful values

marcobergman, 2020-12-05:
Route Position Bearing: The Console XTE, BRG and RNG are all unusable in this mode. I understand the XTE sent in the NMEA message is 0, but here, I still want to see it. BRG should not be capped to the Max Error Angle but always give me a heading to manually steer. And RNG should be range to next waypoint, that is, when I'm expected to make a heading change. This is all for situational awareness. 

Raspberry Pi 4 version 10 buster
OpenCPN 5.2.4
autopilot_route_pi-0.4.0.17-raspbian-armhf-10-buster-armhf/ from cloudsmith.io
also happens when compiled from source
testing method: https://github.com/marcobergman/ais_simulation

marcobergman, 2021-12-20:
Forget about the BRG, that is now clear.

RNG also works in 0.5, but normal range (second figure in RNG field) is always greater than actual range, does not seem to be correct.

XTE is understood to be set to 0 in the APB message for Waypoint Bearing and Range Position Bearing, but still it would be useful to have the actual distance to track correctly shown in the console for situational awareness purposes.

rgleason, 2023-01-22:
Marco is this fixed now?    IF so, please close it!

marcobergman, 2023-01-23:
No all these points have not been dealt with. I might have split them into separate issues, though. Is it wrong to leave this one open?

[image]

[image]

rgleason, 2023-01-23:
No, it ishould be left open. Thanks.