seandepagnier/autopilot_route_pi #14: AP_Route:  Updating it, have some questions

rgleason, 2021-11-07:
Sean,
1. AP_Route does not have a toolbar icon. Is that correct?
2. Access to control the settings is through the Preferences in the PImanager.
3. Does the Preferences window have to stay open?
4. Do I have to set an active route? Could I set and active route and run Shipdriver to simulate use?
5. How would I know if it is working
6. How would you suggest that I test it to see if it is working properly (running a VDR file doesn't really work, nor does it seem like shipdriver would work.)  Is the only way to test it is in actual use?
7. In the nmea window would I see ECAP sentences?
Thanks.

seandepagnier, 2021-11-07:
1&2) yes
3) no
4)  yes, I dont know
5)  It will how the active route dialog which is different from the builtin one
6)  It outputs APB messages typically (see preferences) so the messages are output via the log.   It should work from VDR
7)  Normally no

rgleason, 2021-11-16:
Thanks Sean, perhaps that information will help. What I gather from your answers is that running a VDR file with an active route should work and show a different looking route console. 
"It outputs APB messages via the log "(the opencpn.log file?)

rgleason, 2022-06-17:
Sean, I believe I have gotten your version of the plugin updated to Testplugin Frontend2 (TP FE2) and it seems to be working for most OS, I just have to fix MacOS, as we are missing a standard file in frontend.

The  branch is master-sean   version is 0.4.6.2  and the tarballs are here for PIM "Import Plugin" testing.
https://cloudsmith.io/~opencpn/repos/autopilot-route-alpha/packages/?q=0.4.6.2+tarball
https://github.com/rgleason/autopilot_route_pi/tree/master-sean
[image]

[image]
[image]

rgleason, 2022-06-18:
Volunteer Testers. Please test autopilot_route_pi  v0.4.6.3 by downloading appropriate OS tarballs and installing using PIM "Import Plugin", then install it, then enable.
Note that there is no Toolbar Icon for this plugin, Settings are accessed through Preferences.
https://cloudsmith.io/~opencpn/repos/autopilot-route-alpha/packages/?q=0.4.6.3+tarball

Please report in the test spreadsheet.

Also see this Issue for menu images.
https://github.com/seandepagnier/autopilot_route_pi/issues/14#issuecomment-1159307397

rgleason, 2022-06-18:
Autopilot_route 0.4.6.3 has been added to the PIM Beta Catalog Please test.
Note that there is no Toolbar Icon for this [plugin](https://www.cruisersforum.com/forums/tags/plugin.html), Settings are accessed through Preferences.

closed 2023-01-18 by seandepagnier