seandepagnier/autopilot_route_pi #17: Standard XTE: BRG indicator incorrect values

marcobergman, 2021-12-02:
It is unclear what the BRG indicator gives in Standard XTE mode, but it does not seem to be the expected Bearing To Active Waypoint.

seandepagnier, 2021-12-02:
It is supposed to be...   I changed the code yesterday for rendering maybe it helped?

marcobergman, 2021-12-20:
Yes, some incremental insight here. It seems it shows the _track bearing_, which I have come to assume to be correct. Track bearing + XTE seems to be two components that could make an autopilot decide what to steer to get back on track. OpenCPN standard behavior shows the Bearing to Waypoint here, though. Whereas this might be handy, I believe it to be incorrect. If you agree, this issue can be closed.

rgleason, 2022-06-14:
Sean, I believe I have managed to take your current version and update it for the current frontend2 so that it can be build for all the OS (not tested of course).

I am not sure how to do the next move to make a good PR back to your repository and get my _rgleason/autopilot_route_pi_  repository on track.

I currently have the code in a local repository  _autopilot_route_sean_  can you offer any suggestions as to how to do this properly?

Should I simply copy those _autopilot_route_sean_  local files over onto my local _rgleason/autopilot_route_pi_ files?
Then make a new commit and try to push it up to my remote repository on github?   ...and use push --force if necessary?

I would like to get this on track and have autopilot_route using your code and be available widely on OS.

rgleason, 2022-06-14:
PS I've built it for windows and been able to install it. I haven't really tested it, but marco has done a lot of that.

closed 2022-06-16 by seandepagnier