seandepagnier/autopilot_route_pi #19: Disabling plugin terminates OpenCPN

marcobergman, 2021-12-02:
Both 0.5 on Raspbian, and 0.4.1.0 on Windows.

Side note: When I go test this plugin at the boat, I'd like to be able to revert to the stock OpenCPN route facility. To have to restart OpenCPN and possibly have lost track and route data is a bit too involved.

seandepagnier, 2021-12-05:
I fixed in git.

closed 2021-12-05 by seandepagnier

marcobergman, 2021-12-05:
It seems to be a bit better, however I can still make opencpn crash by subsequently enabling and disabling the plugin, activating and deactiving a random route in between. Takes only a few cycles. This is on raspbian 10 nbuster with opencpn 5.2.4. I compiled it on ubuntu today again, there is did crash upon disabling. 

I have tried enabling and disabling without activing a route and then it does not crash.

marcobergman, 2022-07-08:
Still happens under windows, the version 0.4.7.0 that was published in the beta plugin manager catalog. It happens upon clicking the OK button in the Options screen, after the autopilot_route plugin enabled checkbox was unchecked. This happens both then there is a route active and when no route is active. 

Disabling the plugin is important because in this phase, one might want to fall back to the OpenCPN stock route facility. With OpenCPN crashing, testing on the water becomes less inviting.