seandepagnier/autopilot_route_pi #26: Not for the Master Catalog

rgleason, 2023-01-15:
After a day of testing with Marco Bergman's excellent ais-simulation tool, and many hours of building and trying to get this plugin "undiverged" from Sean's version, 
I have concluded that without some major improvements, I will not be adding it to the Master Catalog.  It may stay in the Beta Catalog for awhile so others can play with and test it.

Sorry, but there have to be some standards and this plugin will just aggravate users because it does not work properly.

See this thread for more details  https://github.com/marcobergman/ais_simulation/issues/10

rgleason, 2023-01-18:
Thank you Sean!!  Much appreciated. Let me check our testing.

rgleason, 2023-01-18:
Sean these are what I can glean from my testing with Marco here, we will test again if you would like:

Can the console be improved as compared to the OpenCPn console? It would be very useful to be able to adjust the text size and color, etc
See
https://github.com/marcobergman/ais_simulation/issues/10#issuecomment-1383012407
Marco says the linux version has a better looking console.  For me, the text is way to small I need to be able to set it.
https://github.com/marcobergman/ais_simulation/issues/10#issuecomment-1383133392

In the 7th &8th (last)  screenshoot in this Post there are jogs in the route, where the boat just decides to head south briefly and then turn back to a better course.  This can't happen in the real world.

Also I found it was easy to crash OpenCPN by deleting a track created in this plugin, while the route was running.

Some problem where many extra tracks get created which slow everything down and sometimes crash OpenCPN.   The on reboot all the tracks have to be deleted.

seandepagnier, 2023-01-19:
It now appears to continue to follow the deleted route and not crash.   If you can test again and try to make it crash let me know.

rgleason, 2023-01-19:
Sean last night I tested a built version with ais-simulator and while I just tested the mode for "Route Position Bearing" by Distance.
and it works well.  It works better at speed 10, than speed 20, and worse at speed 60, it wobbles, etc. but few boats go that fast.

[image]
[image]

I will test the other modes next.  I am working towards improving the TestPlugin FE2 plugin distribution so that we have the right tarballs for all os. Alec and dave have been helping with review. I hope Jon and I can get this fixed to be able to push this to Beta.

rgleason, 2023-01-19:
Route Position Bearing By Time
This one seems to vascillate in direction quite a bit, no matter how "speedup" is set.  The frequency is greater but the amplitude is less the slower the boat is going.  Also at the end the boat kind of failed because it went back and forth 180 degrees and then did a small turn and left the route on some weird direction.
[image]

 
[image]

[image]

[image]
[image]

rgleason, 2023-01-19:
I am finding that Waypoint Route and  Standard XTE  seem to stop or hang. I have closed Opencpn several times and reopened to do this test. 

Next I took the dll and pdb files (relanddebinfo)  and copied them into the normal %localappdata%/opencon/plugins  and start VS2022 to run Opencpn and the plugin.   When this was done, Nmea data messages started and ais-simulator opened. The plugin set on Waypoint or Standard XTE  and ais-simulator started.  The movement of the after its initial turn, stopped immediately with both modes.

It appears that these modes do not work in Windows.

Here are some screenshots. 
[image]
[image]
[image]

rgleason, 2023-01-21:
Sean and Marco,
I have taken Sean's fixes and improvements, tested them (above) and then updated the TP FE2 to Jon Gough's version 1.0.223 which has some small changes. 

I had hopes of building all the OS tarballs and metadata for O5.7.1 (and O5.8) but that is not a trivial change to the frontend, so we will have to wait for that until an updated version of TP FE2 comes out.  However we have identified what needs to be done to TP FE2 here in the [TP Template Table](https://github.com/jongough/testplugin_pi/issues/284) to have a plugin for each OS.  Unfortunately all of Sean's plugins (and the others) will need to be updated to this future version of TP FE2 to have all the plugins needed for O5.7.1 (O5.8) in PIM(plugin manager). 

Meanwhile, we will have to get by using the PIM Beta Catalog and the distribution that TP FE2 currently has, so that we can get all features working.  I will push a new minor "tweak" version autopilot_route_pi v0.4.8.2  to the Beta Catalog today and then make a PR to Sean's repository.  It is currently building in circleci and pushing the tarballs and metadata to Cloudsmith.

rgleason, 2023-01-21:
I've made the PR to plugins Beta branch  https://github.com/OpenCPN/plugins/pull/803 which bdbcat will probably accept later today.
and have made a PR to Sean's autopilot_route_pi for version 0.4.8.2

marcobergman, 2023-01-23:
I clarified some issues but closed 10 and 11, to keep focus on issues that prevent the move to master.

rgleason, 2023-01-23:
You should reopen them later I think

rgleason, 2024-12-16:
@marcobergman   Are we still having troubles with Sean's AP, or can this be closed now?

marcobergman, 2024-12-17:
No, plugin from the master catalog has been in full swing over the previous 2 seasons. I'm happy as Larry with it. Please do close.

closed 2024-12-17 by rgleason