seandepagnier/autopilot_route_pi #28: RFE: Console Improvement

rgleason, 2023-01-21:
I realize now that the Console is independently created by Autopilot_route  (If I have this wrong, why is the console so different from the one that is used in OpenCPN main program?).

There are features I really like: 
1. Ability to select nmea0183 parameters to show.
2. Ability to move around the screen rather than being attached in one place.
3. Ability to adjust width of the console.
4. Ability to adjust the height of the console.

It would be wonderful to be able to
1. Adjust the Console value font, font size, font color, font transparency
2. Adjust the Console legend font, font size, font color, font transparency.
3. Adjust the Console background color and transparency.

One of the reasons I need 1&2 is because of a High Resolution Display.

[image]

marcobergman, 2023-01-23:
Actually, the 'stock' route console font can be adjusted with the OpenCPN's User Interface Options; I'd suggest making the autopilot_route plugin behave in the same way:

[image]

rgleason, 2023-01-23:
Yes !!!
It would also be wonderful to be able to change the background, including transparency.