seandepagnier/autopilot_route_pi #36: OpenCPN crashes after plugin is disabled

marcobergman, 2023-07-03:
To replicate:

- Enable plugin
- create and activate a route
- deactivate the route
- disable the plugin and click ok
- OpenCPN crashes

Log attached.

Windows OpenCPN 5.8.4; plugin 0.4.9.0 (from the master plugin catalog)

[opencpn.log](https://github.com/seandepagnier/autopilot_route_pi/files/11941284/opencpn.log)

rgleason, 2023-07-03:
marco would you please check version [0.4.](https://github.com/rgleason/autopilot_route_pi/commit/da39ff91e5603cfd2b9205ffcf6c2dac774e97eb)  which is the current version in the PIM master catalog.  Thanks.

rgleason, 2023-07-04:
Marco,  Don't bother with that. Version Autopilot_route 0.4.13 [branch master](https://github.com/rgleason/autopilot_route_pi/tree/master) Android fails. Prod 06/14/2023 - This is not working Preferences does not

I have to do some work on it.   I thought it tested out ok, but I was doing a lot of plugins and might have missed something.

marcobergman, 2023-07-04:
I can't find 0.4.13 in the master catalog for windows.

rgleason, 2023-07-04:
And you updated the catalog probably. Perhaps PIM has identified it as not compatible. I need to look into this further and fix the pi. Sorry

rgleason, 2023-10-15:
I believe this has been fixed?  If so, please close Marco.  Plugin has been updated and newest version is now in PIM master catalog

marcobergman, 2023-10-27:
I'm afraid not. Plugin 0.4.30.0 from the master catalog still has the problem. Easy to replicate here: enable the plugin, create a simple route, enable the route, disable the route, disable the plugin (uncheck the enabled checkbox and hit ok) --> OpenCPN stops. Last lines;

```
11:02:50.879 MESSAGE plugin_loader.cpp:592 PluginLoader: Deactivating PlugIn: C:\Users\Marco Bergman\AppData\Local\opencpn\plugins\autopilot_route_pi.dll
11:02:52.076 MESSAGE chcanv.cpp:2398 Metrics:  m_display_size_mm: 476     g_Platform->getDisplaySize():  1920:1080   
11:02:52.076 MESSAGE chcanv.cpp:2403 wxDisplaySize(): 1920 1080
11:02:52.123 MESSAGE glChartCanvas.cpp:554 BuildFBO 3
11:02:52.139 MESSAGE glChartCanvas.cpp:554 BuildFBO 3
```

rgleason, 2023-10-27:
@bdbcat can you take a look at this bug for us? Thanks rick.

marcobergman, 2023-11-08:
Rick I'm updating my boat systems and now trying to find the autopilot_route plugin in the master catalog of the ubuntu-x86_64:18.04 PIM. Can't see it. Should it be there? Thx!
(same for weatherrouting and pypilot plugin, but if I have the answer for this one, I can probably extrapolate)

bdbcat, 2023-11-09:
marco...
We do not officially support ubuntu 18.04 in PIM.  Bu there are some legacy plugins available for this platform.
You may be able to load plugins if you shift your target system flavor to "debian-x86_64:10"
Settings->Plugins->Settings..
Please let me know how this goes.

rgleason, 2023-11-09:
Thanks Dave, Hope it works Marco, as we'd really like to allow you to test this plugin further in its new configuration.  I believe Dave's suggestion should work.

marcobergman, 2023-11-09:
Thanks for the suggestion, but after plugin installation, opencpn encounters errors during plugin startup, telling me the plugin will be uninstalled again. I guess I have to compile them from github, like I have done many times in the past; the question is now what repositories to choose because the version straight from Sean does compile, but gives me a segment fault.

marcobergman, 2023-11-09:
I'm trying to compile pypilot_pi from https://github.com/bdbcat/pypilot_pi, but at cmake .. I get this:

```
-- CMakeLists: Adding target link libraries to pypilot_pi
CMake Error at CMakeLists.txt:233 (add_subdirectory):
  add_subdirectory given source "opencpn-libs/api-17" which is not an
  existing directory.

CMake Error at CMakeLists.txt:236 (add_subdirectory):
  add_subdirectory given source "opencpn-libs/tinyxml" which is not an
  existing directory.

CMake Error at CMakeLists.txt:239 (add_subdirectory):
  add_subdirectory given source "opencpn-libs/jsoncpp" which is not an
  existing directory.

CMake Error at CMakeLists.txt:242 (add_subdirectory):
  add_subdirectory given source "opencpn-libs/wxJSON" which is not an
  existing directory.

CMake Error at CMakeLists.txt:245 (add_subdirectory):
  add_subdirectory given source "opencpn-libs/plugin_dc" which is not an
  existing directory.
```

Also, it plans to install to /usr/local, but the OpenCPN from the Ubuntu ppa does not search there.

```
-- PluginInstall: *** Will install to /usr/local  ***
```

```
23:08:41.949 MESSAGE plugin_loader.cpp:246 PluginLoader: loading plugins from /home/marcb/.local/lib/opencpn;/usr/lib/opencpn
```

rgleason, 2023-11-10:
@bdbcat  @marcobergman 
Marco, the problem is due to the local submodule opencpn-libs needs to be updated or initialized.  There are two places that you can refer to to do this. One is the readme in the sub-module opencpn-libs directory and the other one is unded TP Isssues
https://github.com/OpenCPN/opencpn-libs/blob/main/README.md
https://github.com/jongough/testplugin_pi/issues/342  scroll down to opencpn-libs  and there are a number of links and after reading you can pick one of them to follow.   What I think I would do is to simply remove the local opencpn-libs directory and then reestablish it using the git commands. 

Note: Dave had told me to stop updating opencpn-libs because he and Leamas were busy changing this common source. I had hoped that Dave would advise re your question, and he may still do so, but I am just giving you what I would do because the plugin is not building right now.  

Hope this helps.

marcobergman, 2023-11-10:
Thx for this. I realise I'm putting unneeded burden on you guys sticking to an ancient linux version. So I'm upgrading my hardware now to upgrade linux. There is a practical limit to reusing old hardware.

rgleason, 2023-11-11:
Added the info here in the newer version of the manual
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:installation:ubuntuppa

Thanks.

marcobergman, 2023-11-11:
Thx Rick! I managed to get 20.04 (an LTS version) installed on my old boat laptop; there I could install the pypilot plugin. 

Installing 22.04 (all other things the same), the pypilot plugin is rejected after login. I have two clean VM's now where I can replicate it. I'm not sure where I can look for logging to provide more details on this. I have no idea how often 22.04 is used to run opencpn and the pypilot plugin, other than by me, so I don't want to take up bandwidth for it ;-)

[image]

bdbcat, 2023-11-12:
On the 22.04 fail:
Have a look at the OCPN logfile, noting anything related to pypilot.
You will probably find a "missing linker symbol" message, of some sort.

marcobergman, 2023-11-12:
Here we go:

4:23:11.973 MESSAGE update_mgr.cpp:222 Selected update: <span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>
14:23:11.980 MESSAGE pluginmanager.cpp:572 Installing <span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>
14:23:11.980 MESSAGE plugin_handler.cpp:1248 Installing <span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>_pi-0.40.30.0-ubuntu-x86_64-22.04-jammy.tar.gz from local cache
14:23:11.982 MESSAGE plugin_handler.cpp:712 archive finish write error: Can&apos;t restore time
14:23:11.982 MESSAGE plugin_handler.cpp:712 archive finish write error: Can&apos;t restore time
14:23:12.007 MESSAGE plugin_loader.cpp:1143 Plugin is compatible by elf library scan: true
14:23:12.016 MESSAGE plugin_loader.cpp:246 PluginLoader: loading plugins from /home/marcb/.local/lib/opencpn;/usr/lib/opencpn
14:23:12.016 MESSAGE plugin_loader.cpp:162 Using LD_LIBRARY_PATH: /home/marcb/.local/lib/opencpn:/usr/lib/opencpn:/home/marcb/.local/lib/opencpn:/usr/lib/opencpn:/home/marcb/.local/lib/opencpn:/usr/lib/opencpn
14:23:12.016 MESSAGE plugin_loader.cpp:194 Using PATH: /home/marcb/.local/bin:/usr/bin:/home/marcb/.local/bin:/usr/bin:/home/marcb/.local/bin:/usr/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
14:23:12.016 MESSAGE plugin_loader.cpp:251 Loading plugins from dir: /home/marcb/.local/lib/opencpn
14:23:12.016 MESSAGE plugin_loader.cpp:444 PluginLoader searching for PlugIns in location /home/marcb/.local/lib/opencpn
14:23:12.017 MESSAGE plugin_loader.cpp:482 Found 1 candidates
14:23:12.017 MESSAGE plugin_loader.cpp:271 Checking plugin candidate: /home/marcb/.local/lib/opencpn/lib<span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>_pi.so
14:23:12.017 MESSAGE plugin_loader.cpp:336 Checking plugin compatibility: /home/marcb/.local/lib/opencpn/lib<span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>_pi.so
14:23:12.017 MESSAGE plugin_loader.cpp:1143 Plugin is compatible by elf library scan: true
14:23:12.017 MESSAGE plugin_loader.cpp:1206 PluginLoader: Loading PlugIn: /home/marcb/.local/lib/opencpn/lib<span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>_pi.so
14:23:12.018   ERROR dlunix.cpp:120 Failed to load shared library &apos;/home/marcb/.local/lib/opencpn/lib<span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>_pi.so&apos;: /home/marcb/.local/lib/opencpn/lib<span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>_pi.so: undefined symbol: _ZNK13wxGLCanvasEGL15IsShownOnScreenEv, version WXU_3.2
14:23:12.018 MESSAGE plugin_loader.cpp:1251 Ignoring blacklisted plugin lib<span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>_pi
14:23:12.018 MESSAGE plugin_loader.cpp:1261    PluginLoader: Cannot load library: /home/marcb/.local/lib/opencpn/lib<span style="background-color:#FFFFFF"><font color="#300A24">pypilot</font></span>_pi.so
14:23:12.018 MESSAGE plugin_loader.cpp:251 Loading plugins from dir: /usr/lib/opencpn

bdbcat, 2023-11-12:
Looks like a Wayland problem.
Please try starting OCPN with Wayland disabled.

$GDK_BACKEND=x11 opencpn

marcobergman, 2023-11-12:
That did not do it, and putting WaylandEnable=false in /etc/gdm3/custom.conf did not do it either. Same output in opencpn.log. I do have a new error message, which shows up when I click pypilot in the plugin list. It causes opencpn to abort:

`(opencpn:24745): Gtk-CRITICAL **: 14:22:58.212: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar`

bdbcat, 2023-11-12:
I'll try some local builds on 22.04, and see if I can isolate this.

bdbcat, 2023-11-13:
Marco...
How did you install OpenCPN on 22.04?  

Dave

marcobergman, 2023-11-13:
_Autopilot_route_ plugin is now fine, it's the _pypilot_ plugin that is the problem on 22.04. I'm guilty of introducing the confusion in [this post](https://github.com/seandepagnier/autopilot_route_pi/issues/36#issuecomment-1806923430); I've now created the issue under the repository of the pypilot plugin: https://github.com/pypilot/pypilot_pi/issues/37. Apologies for any confusion.

The original issue (crashing ocpn after disable) is still actual.

rgleason, 2023-11-13:
marco can we please close this then?  you need to close it you are the original poster OP

marcobergman, 2023-11-13:
Yes, I'm afraid so: The original issue (crashing ocpn after disable) is still actual.

Just checked it: 0.4.30 on Windows still crashes OpenCPN, and 0.4.30 also crashes OpenCPN on 20.04 and on 22.04. Same scenario to replicate, last line in the log:

```
22:05:16.151 MESSAGE plugin_loader.cpp:592 PluginLoader: Deactivating PlugIn: /home/marcb/.local/lib/opencpn/libautopilot_route_pi.so
```

bdbcat, 2023-11-14:
Marco...
Fault corrected in newly published plugin.  v0.4.30.3
Please confirm and close if appropriate.

marcobergman, 2023-11-14:
Thx, that was it!

Tested on Windows, Ubuntu 20.04 and Ubuntu 22.04

closed 2023-11-14 by marcobergman

rgleason, 2023-11-14:
Thanks Dave and Marco.

rgleason, 2023-11-14:
Should this PI be updated now in PIM?

marcobergman, 2023-11-14:
Thank you as well, Rick. I think it already is in the master catalog; that's where I tested it from.