seandepagnier/autopilot_route_pi #37: When intercepting a route, boat lurks when reaching guide radius

marcobergman, 2023-11-11:
When intercepting a route, the boat maintains max-error-angle until the moment it is the guide radius away from the route. At that moment it suddenly lurks to a heading perpendicular to the route, no longer maintaining max-error-angle. 

[image]

This seems to be caused by the change in the latest commit of jan 18th:

[image]

Please remove this; I don't want my boat to make this type of movements. This was observed in the 0.4.30 version of the plugin manager on OpenCPN 5.8.4, Ubuntu 20.04.

marcobergman, 2024-03-16:
It still occurs in 0.4.31, and by now I have also encountered this in real life under sail, with very unpleasant effects.

marcobergman, 2024-03-16:
[image]

seandepagnier, 2024-03-20:
I have this issue in mind,  sorry I am focused on other projects but I will certainly look into it and do some sailing tests soon myself.

seandepagnier, 2024-04-03:
I had made this change a while ago but it did not get committed..    hopefully it is fixed now

marcobergman, 2024-04-06:
Looks good, I'll pull it through the simulation later. What's the extra
functionality added?

marcobergman, 2024-04-09:
Indeed, the original 'lurking' problem seems to be solved. The functionality of the newly introduced 'Intersect' checkbox is not clear, though. When it is 'off', the route is intercepted with MaxErrorAngle (orange track). When it is 'on', the interception angle seems to be higher (yellow track). I've seen the code but could not explain it in words. Otherwise, I'd suggest the intersection checkbox to be 'off' by default.

[image]